sparkfun/APDS-9960_RGB_and_Gesture_Sensor #3: Arduino uno with apds-9960

maosuyun2009, 2015-06-03:
Hi, When I connect APDS-9960 to my arduino uno and run GestureTest example, In init process, I always get 0x9F, when read value from APDS9960_ID register, what's wrong with it

ShawnHymel, 2015-06-03:
The ID should return 0xAB or 0x9C. How are you powering the APDS-9960 and which GPIO did you connect it to? The APDS-9960 is only rated for 3.3V.

maosuyun2009, 2015-06-03:
I connect A4 to SDA, A5 to SCL. VDD to 3.3v

ShawnHymel, 2015-06-03:
That sounds like it will work, considering the I2C lines are pulled up to 3.3V. There might be a new version of the chip that came out. I will pull a board from our production and see what ID reads for me. In the meantime,  you could comment out the lines in the init() function of the library that checks the ID to see if that still works for you.

maosuyun2009, 2015-06-05:
Hi, I comment out the lines in the init() function of the library, It's work perfect. I want to use APDS9960 in my linux system, where can i find apds9960 linux driver ?

ShawnHymel, 2015-06-08:
Great! I still have not had a chance to pull a new APDS9960 yet, but as soon as I do to verify the new address, I will update the library. Consider this the official workaround for now :)

There is no Linux driver that I am aware of. It would have to be ported to regular C and compiled for Linux. If you happen to find/make one, please let me know! Sadly, porting one isn't something I have time for right now :(

maosuyun2009, 2015-06-10:
Hi,
    I have ported this libray to my ar9331 linux board, Because ar9331 don't have i2c controller, so I use gpio to simulate i2c. But I can't read any register value except APDS9960_ID register(it's value is 0x9F). I have used my logic analyzer checked gpio output data, it's right. But Arduino uno can read register value correctly. The only differenc between Arduino and ar9331 is IO ouput voltage, ar9331 IO output voltage is 2.7v and arduino is 3.3v, Does it matter?

ShawnHymel, 2015-06-10:
According to the datasheet, the minimum I2C voltage for logic high is 1.26 V, so you should be fine. Remember that I2C lines are supposed to be open drain, so you might be affecting the lines with push-pull gpio. Can you configure them for open drain? Check out this post for using I2C from the AR9331: https://forum.openwrt.org/viewtopic.php?pid=166881#p166881

maosuyun2009, 2015-06-16:
Hi, I run ProximitySensor And GestureTest example in my arduino board, But I don't know why apds9960 continues send interrupt signal to arduino, Event though proximity is 0. Because of interrupt continues happened, when I hand over apds9960, I can't seen UP/DOWN or LEFT/RIGHT print

maosuyun2009, 2015-06-16:
Hi, I'm sorry bother you again. Ignore my previous question. I run ProximitySensor example code in my arduino uno and arduino pro board. arduino uno run perfectly, But arduino pro always read fifo data in while(1) loop. It seems APDS9960_GSTATUS register equals APDS9960_GVALID always true

ShawnHymel, 2015-06-16:
With the ProximitySensor example, the gesture recognition engine is never started (as with apds.enableGestureSensor(true) in the GestureTest example). I'm not sure what the behavior of GSTATUS and GVALID would be without setting the Gesture Enable bit. Can I ask why you are trying to read the gesture registers in the ProximitySensor example?

maosuyun2009, 2015-06-17:
Hi, I don't clearly explain my question. Exactly, When I run GestureTest example code in my Arduino Pro board, I have same problem like this https://forum.sparkfun.com/viewtopic.php?f=14&t=40026

ShawnHymel, 2015-06-17:
That's starting to sound like it might be a bad APDS-9960. Do you have a 3.3V Arduino or a logic level converter you could try? Otherwise, I recommend calling our customer service to see if you can get a replacement.

ShawnHymel, 2016-06-21:
Closing issue after no activity.

closed 2016-06-21 by ShawnHymel