sparkfun/APDS-9960_RGB_and_Gesture_Sensor #6: Gesture recognition fails if one diode pair is deactivated

HoLo85, 2016-04-06:
Hi, 
there is a problem in the gestur recognition if you deactivate one pair of the photodiodes via the
"Gesture Configuration Three Register (0xAA)" register. 
If one gesture dimension is switched off the complete recognition does not work anymore. It always returns NONE.
A quick fix seems to be by commenting out the lines 816-820 and changing the AND (&&) in line 803 and 836 into a OR (||).

ShawnHymel, 2016-06-21:
Thanks for bringing this up. If I get a chance, I will test this out on some hardware. I am currently swamped with other testing and tutorials at the moment.

MartyMacGyver, 2017-02-14:
Are those line refs correct for `Libraries/Arduino/src/SparkFun_APDS9960.cpp`? Because they don't seem to correspond to the current code, and it's unchanged from 2 years ago.

HoLo85, 2017-02-23:
Hi, sry for the late response. Looks like the code i had was a bit diffrent to the current code in the repo.
But the errors if described are still in there.
Comment this lines out (805-809):
`/* If one of the _first values is 0, then there is no good data */
        if( (u_first == 0) || (d_first == 0) || \
            (l_first == 0) || (r_first == 0) ) {
            
            return false;
}`
and change these:
`if( (gesture_data_.u_data[i] > GESTURE_THRESHOLD_OUT) &&
                (gesture_data_.d_data[i] > GESTURE_THRESHOLD_OUT) &&
                (gesture_data_.l_data[i] > GESTURE_THRESHOLD_OUT) &&
(gesture_data_.r_data[i] > GESTURE_THRESHOLD_OUT) ) {`
to
`if( (gesture_data_.u_data[i] > GESTURE_THRESHOLD_OUT) &&
                (gesture_data_.d_data[i] > GESTURE_THRESHOLD_OUT) ||
                (gesture_data_.l_data[i] > GESTURE_THRESHOLD_OUT) &&
                (gesture_data_.r_data[i] > GESTURE_THRESHOLD_OUT) ) {`
in line 791 and 823.

AndyEngland521, 2020-06-04:
Thanks for you help all! Adding these changes in v1.4.3

closed 2020-06-04 by AndyEngland521