sparkfun/APDS-9960_RGB_and_Gesture_Sensor #8: Color Sensing

vlast3k, 2016-07-02:
Can you please give me a hint, what the color sense (RGB feature) does?
I tried it but somehow the results aren't consistent
It always shows almost equal amounts of R G and B about 1/3 of the total ambient light
I tried to put the sensor in front of something Blue or Green or Red, but nothing changed too drasticaly.

Is this the feature or does it do something else ?

ShawnHymel, 2016-07-03:
It does sense amounts of red, green, and blue, but I find that it is not as accurate as some other dedicated RGB sensors. I recommend trying it with a colored light source, like a blue LED.

vlast3k, 2016-07-03:
ahh, i also suspected it would work better with light sources, thanks :)

closed 2016-07-03 by vlast3k