sparkfun/APDS-9960_RGB_and_Gesture_Sensor #10: Sensor will NOT work if logiclevel converter used

igittigitt, 2017-02-12:
Very strange, i bought such a sensor and also a bidirectional converter. But if i use this, the demo gesture sketch tells me that the sensor could not be initialized. I measured the voltages and can confirm the levelconverter works correct. On APDS-Side it uses 3.3V for SCA,SCL and also INT (i know it's not necessary, but so what the board has 4 converters on it) and on the Arduino Uno side 5V for logic level.

Now, after playing around for a while a decided to test the signals (! NOT Vcc, for shure!) without the converter, hence i know it possible kill my sensor... But for some reason it then works! I'm confused. Checked everything double, test again, but it is that way: using a converter -> no chance. without it -> all right.

Can somebody clearify this? The sensor is the simple one, without any extra logic on board. Just the cap and the pullups.

ShawnHymel, 2017-02-13:
What kind of logic level converter are you using? I2C has problems with them sometimes. 

Also, because I2C is open drain, you can just connect VCC to 3.3V from the Arduino and connect SDA and SCL to your I2C pins. SCL and SDA will be driven LOW but never HIGH (open drain). The 4.7k pullups on the APDS-9960 breakout will ensure that SCL and SDA will be nominally pulled to 3.3V (not 5), which acts as a sort of level converter. 

You likely don't need a level converter on INT, as it's pulled high to 3.3V as well, which is good enough for logic level high for your 5V TTL on your Arduino for GPIO input.

igittigitt, 2017-02-21:
Thanks a lot. This all makes sense to mee! But i think one should strictly void to use internal pullup of inputs (activated by the "_PULLUP" macro) and only use those external pullup resistors.
It not really explains why using an, even unnecessary, levelshift will break communication totally... i'm using this kind of convert: http://i.ebayimg.com/images/g/wK0AAOSw8aNXGkGD/s-l1600.jpg

ShawnHymel, 2017-02-21:
That's a similar converter to the one SparkFun sells, so it's good for general I/O, SPI, and UART (just not at super high speeds). And yes, with I2C, external pullups are the way to go. Did everything work out for you?

RupinC, 2018-04-04:
I am having a similar problem. Did this scheme of wiring work?

loricrotser, 2019-11-01:
I am closing this issue since it hasn't been touched in over a year. If you need further technical assistance, feel free to visit the [SparkFun forums](https://forum.sparkfun.com/index.php) and we'll see what we can do to help you out. Thank you!

closed 2019-11-01 by loricrotser