sparkfun/APDS-9960_RGB_and_Gesture_Sensor #11: Sensor not working with a glass cover above it

nikilrao, 2017-02-23:
I'm using a glass cover (90% IR transmittance) on top of the sensor. Because of this there's always data in the FIFO and it's stuck in this if statement - 

```
if( (gstatus & APDS9960_GVALID) == APDS9960_GVALID ) {
.......}
```

From what I understood, the FIFO becomes empty after the gesture is completed and then it enters this else statement - 

```
else {
    
            /* Determine best guessed gesture and clean up */
            delay(FIFO_PAUSE_TIME);
            decodeGesture();
            motion = gesture_motion_;
#if DEBUG
            Serial.print("END: ");
            Serial.println(gesture_motion_);
#endif
            resetGestureParameters();
            return motion;
        }
```

and the final gesture is determined. So the gesture is not being detected because it never enters this else statement. How can I solve this problem? Thanks!

HoLo85, 2017-02-24:
How big is the distance between the cover glass and your sensor?
If its to big the sensor will think there is always something above it.

nikilrao, 2017-02-24:
I've tried almost all distances ranging from right on top of the sensor to a few mm above it.

HoLo85, 2017-02-27:
My problem is that i dont have access to the sensor anymore, so im maybe not a big help.
@nikilrao have you tried putting a foam around the sensor to keep light out from the environment?
That helped me when i worked with the sensor. I also removed the capacitor and put it on the backside of the board, to be able to glue the sensor directly to the cover i used.

lillyagar, 2018-09-07:
Hello @HoLo85 I know its a year later! lol but just wanted to know how foam work for you? I am trying to do the same thing but I'm not having any luck. If possible, could you attach a photo of it, to see an example? Thank you! (:

HoLo85, 2018-12-23:
Hi, sry for the (really) late answer. I stopped working on the project over a year ago and i don't have any access to it anymore. In the end a thin layer of foam around the sensor helped, but you should keep the gap between the sensor and glas below 1 mm.

loricrotser, 2019-10-30:
Hey all, since this hasn't been touched in almost a year, I'm going to close the issue. 

If you need further technical assistance, head over to our [forums](https://forum.sparkfun.com/index.php) and we'll see what we can do to help you out. Thank you!

closed 2019-10-30 by loricrotser