sparkfun/APDS-9960_RGB_and_Gesture_Sensor #17: Gesture and proximity don't work well together

SelaO, 2017-10-11:
I have this in my setup():

 ```
  // Initialize interrupt service routine
  attachInterrupt(0, interruptRoutine, FALLING);

  // Initialize APDS-9960 (configure I2C and initial values)
  if ( apds.init() ) {
    Serial.println(F("APDS-9960 initialization complete"));
  } else {
    Serial.println(F("Something went wrong during APDS-9960 init!"));
  }
  
  // Adjust the Proximity sensor gain
  if ( !apds.setProximityGain(PGAIN_2X) ) {
    Serial.println(F("Something went wrong trying to set PGAIN"));
  }
  
  // Start running the APDS-9960 proximity sensor (no interrupts)
  if ( apds.enableProximitySensor(false) ) {
    Serial.println(F("Proximity sensor is now running"));
  } else {
    Serial.println(F("Something went wrong during sensor init!"));
  }

    // Start running the APDS-9960 gesture sensor engine
  if ( apds.enableGestureSensor(true) ) {
    Serial.println(F("Gesture sensor is now running"));
  } else {
    Serial.println(F("Something went wrong during gesture sensor init!"));
  }
```

And the values from the proximity are either 41 or below 10, but if I remove the `if ( apds.enableGestureSensor(true) ) {...` then it works well with values between 0 and 255. 

How can I make both of those features to work well together?

ShawnHymel, 2017-10-11:
It's been a while since I've used the APDS-9960, but I think I've had issues with these two working together in the past. Because the GenstureSensor part of the library relies on the proximity sensor (it's what triggers the gesture engine to start recording events), getting them to work together at the same time might be difficult. Is there a way where you can have only one operating at a time (e.g. proximity on, when it detects a certain event, turn the gesture engine on and disable the proximity portion)?

SelaO, 2017-10-11:
I need them to work together...

loricrotser, 2019-10-31:
After reading the [datasheet](https://cdn.sparkfun.com/datasheets/Sensors/Proximity/apds9960.pdf), it appears that control of that drive pin is shared between proximity and gesture functionality, so having them work together at the same time is going to be tricky. You'd have to boot multiple engines at once, do your reading, and then get the data from both. I haven't tried it, but you might try the [SparkFun forums](https://forum.sparkfun.com/index.php) to see if someone else has done it. Hope that is helpful!

closed 2019-10-31 by loricrotser