sparkfun/APDS-9960_RGB_and_Gesture_Sensor #29: Esp8266 GestureTest

Oopas, 2020-11-13:
I tried to use the library on esp8266, but it was constantly rebooting....

Oopas, 2020-11-13:
Here's some information

--------------------------------
SparkFun APDS-9960 - GestureTest
--------------------------------
ISR not in IRAM!

User exception (panic/abort/assert)
--------------- CUT HERE FOR EXCEPTION DECODER ---------------

Abort called

>>>stack>>>

ctx: cont
sp: 3ffffef0 end: 3fffffc0 offset: 0000
3ffffef0:  2d2d2d2d 2d2d2d2d 2d2d2d2d 2d2d2d2d  
3fffff00:  000000fe 00000000 00000000 00000000  
3fffff10:  00000000 00000000 00000000 00ff0000  
3fffff20:  5ffffe00 5ffffe00 000a0d00 00000000  
3fffff30:  00000002 00000000 3ffee4f4 40202cb2  
3fffff40:  4010072e 402043d5 ffffffff 40202cc4  
3fffff50:  40201f9c 3ffee4f4 3ffee4f4 4020391d  
3fffff60:  00000000 3ffee4f4 3ffe8631 402021a1  
3fffff70:  3fffdad0 00000020 3ffee4f4 3ffee55c  
3fffff80:  3fffdad0 00000000 3ffee4f4 402039cc  
3fffff90:  3fffdad0 00000000 3ffee4f4 4020109c  
3fffffa0:  feefeffe 00000000 3ffee51c 402028c0  
3fffffb0:  feefeffe feefeffe 3ffe84e8 40101029  
<<<stack<<<

jschoch, 2021-03-16:
you may need to add ICACHE_RAM_ATTR to the isr function...

```
void ICACHE_RAM_ATTR interruptRoutine() {
  isr_flag = 1;
}
```

Bob0707, 2023-09-09:
73 |   attachInterrupt(0, interruptRoutine, FALLING);
95 |     attachInterrupt(0, interruptRoutine, FALLING);
Compilation error: 'interruptRoutine' was not declared in this scope

That is what i get after add ICACHE_RAM_ATTR to the ISR function, can anyone help?